Tympan/Earpiece_BTE-RIC_V1 #1: Error in 7146-01003_FLEX CIRCUIT_Rev-E.PDF

biomurph, 2019-09-17:
It seems that there are incorrect dimensions in the drawing of the flex pcb. The same issue is in the DXF file. It has to do with the width of the area of major bending, and the position of the edges of the major bend area.

The listed width of the major bend area is 2.202mm.
From the bottom edge of the board to the bottom edge of the major bend area is listed as 5.532. 

That makes the overall height of the board 13.266mm, which is 0.016 over the plus/minus tolerance.

In KiCad, I have the board outline laid out to match the DXF import, and I find that the width of the major bend area is more like 2mm, not 2.202mm.
[image]

erk1313, 2019-09-17:
The flex width overall and the flex width at the bend are independent dimensions.  

For example, the flex width at the bend could be anywhere from 2-2.20mm.  If you found the die tends to make things smaller, the toolmaker might make the die 2.2mm wide to account for that.  That doesn't mean the overall width needs to increase to 13.2mm.  It can still remain 13mm and still pass the drawing's tolerance. 

 If it did increase to 2.2mm, then the far side of that section (which is not dimensioned) would be at 5.532+2.2mm=7.732mm, which would be fine, and would also not bump out the overall width

biomurph, 2019-09-17:
OK, thanks for the clarification on that. I will be moving traces and vias around and if I need the 0.2 extra width it may come in handy.

closed 2019-09-17 by biomurph