Tympan/Earpiece_BTE-RIC_V1 #4: Review BTE Rigid Board Design

erk1313, 2019-09-20:
- Check board edges
-Check schematic
-Check BOM

erk1313, 2019-09-20:
Board edges line up great.
[image]

erk1313, 2019-09-20:
Pads for FPC connector were shifted a bit, which makes sense--they needed more room for the board margin. 
[image]
 I revised the part in SW an updated the drawing.  Outlines now match
[image]

erk1313, 2019-09-20:
Pin order of the body cable was different than in my wiring diagram.  Eric will correct the wiring diagram.
[image]

[image]

erk1313, 2019-09-20:
Routing of wires is correct (other than speaker - and speaker + was swapped).  See issue #5

erk1313, 2019-09-20:
Should there be a BOM and a pick and place table?  Created issue #7

biomurph, 2019-09-20:
I had to swap the wire order to ease routing.
Are you are able to make the changes on your end to the slight offset of my layout?

erk1313, 2019-09-20:
Review complete

closed 2019-09-20 by erk1313

erk1313, 2019-09-20:
Yes, corrected it on my end (Issue #9)