Tympan/Earpiece_BTE-RIC_V1 #8: Review BTE Flex design

erk1313, 2019-09-20:
Logging our past communication (Joel + Eric) and adding a few more checks
- [x] Outline
- [x] Distance between mic holes
- [x] Tail length: Note that the tail from Joel's flex outline is a little longer.  That’s because in Solidworks I modeled the minimum tail length.  You were right to make the tail a little longer.  Thank you for paying attention to detail. No change needed
- [ ]Routing
- [ ] Check pin order
- [ ] Check that pads are located on the correct side (front vs back)

erk1313, 2019-09-20:
* Pads for mic are located correctly on the back side
* Checking pin order of the tail

biomurph, 2019-09-20:
- Checking pin order of the tail

Yes, please do. I stared at that for a long time too!

closed 2019-09-20 by erk1313