Tympan/Earpiece_BTE-RIC_V1 #11: Suffix DXF is causing glitches

biomurph, 2019-09-24:
This may not be a critical issue, but the capitalization of DXF is causing a fail in KiCad when I try to import it into the board layout. The fix is to change the suffix from all caps to lower case. Not sure how your getting all caps in all of the suffixes that you've got in this repo, but some softwares might have issues with the all caps.

biomurph, 2020-09-10:
@eyuan-creare I am bringing this up again because my CM is having some hangup opening files.
I tried to change files suffix from caps to lower, but it does not change the repo enough for me to push up to the git.
Right now, I'm sending them zipped files to see if that helps. Can you change the suffixes for the mechanical and pdf files to lower case?

eyuan-creare, 2020-09-10:
@biomurph The file extensions were updated.  Note that I had to add an additional underscore to cause a changed to be detected.